kousheekc/Kinematics #4: Example code is crashing

yaxu, 2022-05-03:
Hi, thanks for sharing this library. I've been having problems getting it to work on my arduino uno.

Below is the example file, just with a few `Serial.println` commands added. It runs fine, until I uncomment the `kin.inverse` line.  Then, I just get junk on the serial console, like:
```
⸮)⸮⸮⸮⸮⸮)⸮⸮⸮⸮
```

Any ideas what could be going wrong? The arduino works fine with other code.

```c
#include "Kinematics.h"
#include "MatrixUtils.h"

#define N 3

void setup() {
    // put your setup code here, to run once:
    Serial.begin(9600);
    Serial.println("hello");

    Kinematics kin(N);
    MatrixUtils mat_utils;
    Serial.println("well.");

    kin.add_joint_axis(0, 0,  1,  4, 0,    0);
    kin.add_joint_axis(0, 0,  0,  0, 1,    0);
    kin.add_joint_axis(0, 0, -1, -6, 0, -0.1);
    Serial.println("added joints.");

    kin.add_initial_end_effector_pose(-1, 0,  0, 0,
                                       0, 1,  0, 6,
                                       0, 0, -1, 2,
                                       0, 0,  0, 1);
    Serial.println("added nose pose.");

    float desired_transform[4][4] = {
        {0, 1,  0,     -5},
        {1, 0,  0,      4},
        {0, 0, -1, 1.6858},
        {0, 0,  0,      1}
    };

    float jac[6][N];
    float jac_t[6][N];
    float AA_t[6][6];
    float A_tA[N][N];
    float pinv[N][6];

    float joint_angles_0[N] = {1.0, 2.5, 3};
    float joint_angles[N];

    // kin.inverse((float*)desired_transform, (float*)jac, (float*)pinv, (float*)jac_t, (float*)AA_t, (float*)A_tA, joint_angles_0, 0.01, 0.001, 20, joint_angles);
    // mat_utils.print_matrix(joint_angles, 1, N, "Joint angles");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

kousheekc, 2022-05-03:
Hi, I'm sorry to inform you that unfortunately, this library does not support boards like the Arduino Uno right now. The inverse kinematics step requires large amounts of matrix manipulations and optimization therefore the memory on a board like the Arduino Uno is just not enough. You could try running the program on an Arduino compatible board that has a larger memory such as the Mega or Due and it should run fine.

In the meanwhile, I am working on certain memory optimization steps which could help resolve the issue.

yaxu, 2022-05-03:
Aha, thanks for your fast and helpful reply @kousheekc! That explains my issue. I'll look for another arduino.