kousheekc/Kinematics #8: Example of 6 axis code or demo?

dalethomas81, 2024-08-12:
Hello - thank you for creating and sharing this library. Would you be willing to share the code you used to create the demo video? I am trying to figure out the screw parameters for the [AR4 robot](https://www.anninrobotics.com/) from Chris Annin and the forward kinematics results are not expected.

Here is what I am using:
  double L1 = 64.2;
  double L2 = 169.77;
  double L3 = 305;
  double L4 = 222.63;
  double L5 = 36.25;

  kin.add_joint_axis(0, 0, 1,  0, 0, 0); // axis 1
  kin.add_joint_axis(0, 1, 0, L1, 0, L2); // axis 2
  kin.add_joint_axis(0, 1, 0, L1, 0, L2+L3); // axis 3
  kin.add_joint_axis(0, 0, 1, L1, 0, L2+L3); // axis 4
  kin.add_joint_axis(0, 1, 0, L1, 0, L2+L3+L4); // axis 5
  kin.add_joint_axis(0, 0, 1, L1, 0, L2+L3+L4); // axis 6

  kin.add_initial_end_effector_pose(  1,  0,  0,  L1,
                                      0,  1,  0,  0,
                                      0,  0,  1,  L2+L3+L4+L5,
                                      0,  0,  0,  1);

Here are the results with J2 at 90 degrees:
0  0  1  967.62
0  1  0  0
-1  0  0  41.37
0  0  0  1

Here are the expected results with J2 at 90 degrees:
0  0  1  628.08
0  1  0  0
-1  0  0  169.77
0  0  0  1

Here is the schematic of the robot:
[image]

dalethomas81, 2024-08-12:
Update: I was able to figure this out after studying screw theory some more. Here are the values I used for anyone that comes across this later:

        double L1 = 64.2;
        double L2 = 169.77;
        double L3 = 305;
        double L4 = 222.63;
        double L5 = 36.25;

        // screw axis expressed in "s-frame" 
        kin.add_joint_axis(0, 0, 1, 0, 0, 0); // axis 1
        kin.add_joint_axis(0, 1, 0, -L2, 0, L1); // axis 2
        kin.add_joint_axis(0, 1, 0, -L2-L3, 0, L1); // axis 3
        kin.add_joint_axis(0, 0, 1, 0, -L1, 0); // axis 4
        kin.add_joint_axis(0, 1, 0, -L2-L3-L4, 0, L1); // axis 5
        kin.add_joint_axis(0, 0, 1, 0, -L1, 0); // axis 6

        kin.add_initial_end_effector_pose(  1, 0, 0, L1,
                                            0, 1, 0, 0,
                                            0, 0, 1, L2 + L3 + L4 + L5,
                                            0, 0, 0, 1);

closed 2024-08-12 by dalethomas81

kousheekc, 2024-08-13:
Hello, glad that you managed to work it out yourself! For other people looking at this issue, I understand the screw theory approach might not seem like the most intuitive but this allows us to follow a more conventional approach to defining joints and links, namely the URDF format. I am working on adding a helper function into the library that automatically parses a URDF file and extracts the joint axes so that users of the library do not need to figure it out themselves.